davidramiro/Marlin-Ai3M-2.0.x #2: Filament In or Filament remove trough the LCD doesnt work.

shaddow501, 2019-12-04:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

davidramiro, 2019-12-04:
Need more information - which version? Latest release or latest commit?

shaddow501, 2019-12-04:
version 2.0.x and downloaded it a few hours ago..

davidramiro, 2019-12-04:
Self-compiled from 2.0.x branch or with a release hex file?

shaddow501, 2019-12-04:
self compiled

davidramiro, 2019-12-04:
Thanks for the info. I recently merged all the commits up to the official 2.0 release. I still have to check everything, thanks for reporting this.

shaddow501, 2019-12-04:
Thanks for your hard work

shaddow501, 2019-12-09:
Do you have any idea where to look for this feature ?
I mean the menu exists but pressing the LCD "buttons" doesnt do anything.. (only beeps lol)

davidramiro, 2019-12-09:
The function is located right here:

https://github.com/davidramiro/Marlin-Ai3M-2.0.x/blob/bugfix-2.0.x/Marlin/src/lcd/anycubic_TFT.cpp#L1119

It's simply not working for the current 2.0, I haven't had the time yet to make the code compatible.

shaddow501, 2019-12-09:
I see that the function is similar to the function of the 1.1.9 version, so it might be related to the touch screen itself that it doesnt recognize the key that is pressed.
So maybe we should try to locate where is the part that deals with the touch itself?

davidramiro, 2019-12-09:
It's exactly the same function, I just need to fix the communication because Marlin 2.0 uses a slightly different syntax. I already did alter the queueing commands but couldn't yet test it.

The touch part is handled solely by the touchscreen itself. It's a closed source firmware running on the display's own MCU. So that has nothing to do with Marlin itself. We just read the incoming command from the display and forward it into Marlin's movement queue.

I will probably look into it this weekend.

shaddow501, 2019-12-09:
If you dont have time to test it and need my help than I will be happy to help

davidramiro, 2019-12-09:
Thank you - for now it would only be great to know if regarding the display, filament in/out is the only function that isn't working or if there's more.

shaddow501, 2019-12-09:
Well since I have installed the BMG extruder I moved back to 1.1.9 since it is hard to insert the filament inside, I need to run to the PC every time and do that remotely, which is difficult to hold the filament inside and than to run to the PC for the command....

belotv, 2019-12-09:
davidamiro,
For the GCODEs you hardcoded, I am not sure you actually store them in memory, it looks like you are missing the PSTR(). The ones you build from sprintf_P seem to be ok.

shaddow501, 2019-12-09:
belotv
this is exactly what I think, I made a modification and now testing

belotv, 2019-12-09:
Let me know if it works. Note that you also have a hardcoded GCODE call in M888 definition.

shaddow501, 2019-12-09:
I did this change but still doesnt work.. 

@@ -166,11 +166,11 @@ void AnycubicTFTClass::StartPrint() {
         SERIAL_ECHOLNPGM("DEBUG: Regular Start");
       #endif
       break;
     case 1:
       // regular sd pause
-      queue.enqueue_now_P("M24"); // unpark nozzle
+      queue.enqueue_now_P(PSTR("M24")); // unpark nozzle
       #ifdef ANYCUBIC_TFT_DEBUG
         SERIAL_ECHOPAIR(" DEBUG: AI3M Pause State: ", ai3m_pause_state);
         SERIAL_EOL();
         SERIAL_ECHOLNPGM("DEBUG: M24 Resume from regular pause");
       #endif
@@ -197,11 +197,11 @@ void AnycubicTFTClass::StartPrint() {
         SERIAL_ECHOLNPGM("DEBUG: Filament Change Flag cleared");
       #endif
       break;
     case 3:
       // paused by filament runout
-      queue.enqueue_now_P("M24"); // unpark nozzle and resume
+      queue.enqueue_now_P(PSTR("M24")); // unpark nozzle and resume
       #ifdef ANYCUBIC_TFT_DEBUG
         SERIAL_ECHOLNPGM("DEBUG: M24 Resume from Filament Runout");
       #endif
       IsParked = false; // clear flags
       ai3m_pause_state = 0;
@@ -254,22 +254,22 @@ void AnycubicTFTClass::PausePrint() {
     } else { // pause caused by filament runout
       #ifdef ANYCUBIC_TFT_DEBUG
         SERIAL_ECHOLNPGM("DEBUG: Filament Runout Pause");
       #endif
       // filament runout, retract and beep
-      queue.enqueue_now_P("G91"); // relative mode
-      queue.enqueue_now_P("G1 E-3 F1800"); // retract 3mm
-      queue.enqueue_now_P("G90"); // absolute mode
+      queue.enqueue_now_P(PSTR("G91")); // relative mode
+      queue.enqueue_now_P(PSTR("G1 E-3 F1800")); // retract 3mm
+      queue.enqueue_now_P(PSTR("G90")); // absolute mode
       buzzer.tone(200, 1567);
       buzzer.tone(200, 1174);
       buzzer.tone(200, 1567);
       buzzer.tone(200, 1174);
       buzzer.tone(2000, 1567);
       #ifdef ANYCUBIC_TFT_DEBUG
         SERIAL_ECHOLNPGM("DEBUG: Filament runout - Retract, beep and park.");
       #endif
-      queue.enqueue_now_P("M25"); // pause print and park nozzle
+      queue.enqueue_now_P(PSTR("M25")); // pause print and park nozzle
       ai3m_pause_state = 3;
       #ifdef ANYCUBIC_TFT_DEBUG
         SERIAL_ECHOLNPGM("DEBUG: M25 sent, parking nozzle");
         SERIAL_ECHOPAIR(" DEBUG: AI3M Pause State: ", ai3m_pause_state);
         SERIAL_EOL();
@@ -311,11 +311,11 @@ void AnycubicTFTClass::StopPrint(){
   TFTstate=ANYCUBIC_TFT_STATE_SDSTOP_REQ;
 }
 
 void AnycubicTFTClass::FilamentChangeResume(){
   // call M108 to break out of M600 pause
-  queue.enqueue_now_P("M108");
+  queue.enqueue_now_P(PSTR("M108"));
   #ifdef ANYCUBIC_TFT_DEBUG
     SERIAL_ECHOLNPGM("DEBUG: M108 Resume called");
   #endif
 
   // remove waiting flags
@@ -337,22 +337,22 @@ void AnycubicTFTClass::FilamentChangePause(){
     SERIAL_ECHOPAIR(" DEBUG: AI3M Pause State: ", ai3m_pause_state);
[... 230 lines cut ...]
@@ -1320,22 +1320,22 @@ void AnycubicTFTClass::GetCommandFromTFT()
                   {
                     s_zoffset=ftostr32(float(CodeValue())/100.0);
                     sprintf_P(value,PSTR("M851 Z"));
                     strcat(value,s_zoffset);
                     queue.enqueue_now_P(value); // Apply Z-Probe offset
-                    queue.enqueue_now_P("M500")); // Save to EEPROM
+                    queue.enqueue_now_P(PSTR("M500")); // Save to EEPROM
                   }
                 #endif
               }
               ANYCUBIC_SERIAL_ENTER();
               break;
             case 32: // A32 clean leveling beep flag
               if(CodeSeen('S')) {
                 #ifdef ANYCUBIC_TFT_DEBUG
                   SERIAL_ECHOLNPGM("TFT Level saving data...");
                 #endif
-                queue.enqueue_now_P("M500\nM420 S1\nG1 Z10 F240\nG1 X0 Y0 F6000");
+                queue.enqueue_now_P(PSTR("M500\nM420 S1\nG1 Z10 F240\nG1 X0 Y0 F6000"));
                 ANYCUBIC_SERIAL_SUCC_START;
                 ANYCUBIC_SERIAL_ENTER();
               }
               break;
             #endif

shaddow501, 2019-12-09:
The dot on the left show what was previous and what is changed.

davidramiro, 2019-12-09:
> davidamiro,
> For the GCODEs you hardcoded, I am not sure you actually store them in memory, it looks like you are missing the PSTR(). The ones you built from sprintf_P seem to be ok.

Thanks, that could very well be. 

I was looking for a replacement for the now obsolete `enqueue_and_echo_commands_P()`, `enqueue_now_P()` seemed to be a decent fit from a quick glance at the comments on queue.cpp. On a second thought, maybe `enqueue_one()` would be more fitting?

Haven't been able to test since I'm still on 1.1.9 because I need to print some stuff, I'll check 2.0 later this week.

>Please explain since I dont know what you mean by: hardcoded GCODE call in M888 definition

He means that the same issue is present in /src/gcode/temperature/M888.cpp. I'll have to make the same adjustments there as well.

shaddow501, 2019-12-09:
Well it is strange since those commands work...

ase 21: // A21 all home
            if((!planner.movesplanned()) && (TFTstate!=ANYCUBIC_TFT_STATE_SDPAUSE) && (TFTstate!=ANYCUBIC_TFT_STATE_SDOUTAGE))
            {
              if(CodeSeen('X')||CodeSeen('Y')||CodeSeen('Z'))
              {
                if(CodeSeen('X')) queue.enqueue_now_P(PSTR("G28 X"));
                if(CodeSeen('Y')) queue.enqueue_now_P(PSTR("G28 Y"));
                if(CodeSeen('Z')) queue.enqueue_now_P(PSTR("G28 Z"));
              }
              else if(CodeSeen('C')) queue.enqueue_now_P(PSTR("G28"));

davidramiro, 2019-12-09:
>Well it is strange since those commands work...

That's good to know. Probably an issue with  `sprintf_P(value,PSTR("G1 X0.1F%i"),movespeed` and the like then.

shaddow501, 2019-12-09:
Maybe it doesnt get the value according to this:

belotv, 2019-12-10:
It seems that functions are actually a bit different.
enqueue_now_P(PGM_P const pgcode)
enqueue_and_echo_command(const char* cmd, bool say_ok=false);
enqueue_one_now(const char* cmd)

shaddow501, can you try to replace the calls to enqueue_now_P that are using sprintf_P results by "enqueue_one_now(" ?

Additionally, for constant GCODE calls, we could take the same approach than vanilla Marlin to save some memory:
const char G91_STR[] PROGMEM = "G91"
and later call it through queue.enqueue_now_P(G91_STR) (it avoids using PSTR)

shaddow501, 2019-12-10:
Dear belotv

I can confirm that doing this change at this part of the code did fix the issue with the X,Y,Z,E moments trough the LCD commands:

` case 22: // A22 move X/Y/Z or extrude
            if((!planner.movesplanned()) && (TFTstate!=ANYCUBIC_TFT_STATE_SDPAUSE) && (TFTstate!=ANYCUBIC_TFT_STATE_SDOUTAGE))
            {
              float coorvalue;
              unsigned int movespeed=0;
              char value[30];
              if(CodeSeen('F')) // Set feedrate
              movespeed = CodeValue();

              queue.enqueue_now_P(PSTR("G91")); // relative coordinates

              if(CodeSeen('X')) // Move in X direction
              {
                coorvalue=CodeValue();
                if((coorvalue<=0.2)&&coorvalue>0) {sprintf_P(value,PSTR("G1 X0.1F%i"),movespeed);}
                else if((coorvalue<=-0.1)&&coorvalue>-1) {sprintf_P(value,PSTR("G1 X-0.1F%i"),movespeed);}
                else {sprintf_P(value,PSTR("G1 X%iF%i"),int(coorvalue),movespeed);}
                queue.enqueue_one_now(value);
              }
              else if(CodeSeen('Y')) // Move in Y direction
              {
                coorvalue=CodeValue();
                if((coorvalue<=0.2)&&coorvalue>0) {sprintf_P(value,PSTR("G1 Y0.1F%i"),movespeed);}
                else if((coorvalue<=-0.1)&&coorvalue>-1) {sprintf_P(value,PSTR("G1 Y-0.1F%i"),movespeed);}
                else {sprintf_P(value,PSTR("G1 Y%iF%i"),int(coorvalue),movespeed);}
                queue.enqueue_one_now(value);
              }
              else if(CodeSeen('Z')) // Move in Z direction
              {
                coorvalue=CodeValue();
                if((coorvalue<=0.2)&&coorvalue>0) {sprintf_P(value,PSTR("G1 Z0.1F%i"),movespeed);}
                else if((coorvalue<=-0.1)&&coorvalue>-1) {sprintf_P(value,PSTR("G1 Z-0.1F%i"),movespeed);}
                else {sprintf_P(value,PSTR("G1 Z%iF%i"),int(coorvalue),movespeed);}
                queue.enqueue_one_now(value);
              }
              else if(CodeSeen('E')) // Extrude
              {
                coorvalue=CodeValue();
                if((coorvalue<=0.2)&&coorvalue>0) {sprintf_P(value,PSTR("G1 E0.1F%i"),movespeed);}
                else if((coorvalue<=-0.1)&&coorvalue>-1) {sprintf_P(value,PSTR("G1 E-0.1F%i"),movespeed);}
                else {sprintf_P(value,PSTR("G1 E%iF500"),int(coorvalue)); }
                queue.enqueue_one_now(value);
              }
              queue.enqueue_now_P(PSTR("G90")); // absolute coordinates
            }
            ANYCUBIC_SERIAL_ENTER();` 

Where should I add this line? const char G91_STR[] PROGMEM = "G91"
 in the start of the file?

davidramiro, 2019-12-10:
> I can confirm that doing this change at this part of the code did fix the issue with the X,Y,Z,E moments trough the LCD commands:

Thanks for testing! I have commited it already.

> Additionally, for constant GCODE calls, we could take the same approach than vanilla Marlin to save some memory:
> const char G91_STR[] PROGMEM = "G91"
> and later call it through queue.enqueue_now_P(G91_STR) (it avoids using PSTR)

Good idea, haven't looked yet into the best practises of Marlin 2. Thanks for your guidance so far!

shaddow501, 2019-12-10:
Lol now I see that I had a typo error moments = movements...

shaddow501, 2019-12-10:
By the way after compiling (with those changes) I get:
Sketch uses 136050 bytes (53%) of program storage space. Maximum is 253952 bytes.
Global variables use 6747 bytes (82%) of dynamic memory, leaving 1445 bytes for local variables. Maximum is 8192 bytes.

1445 bytes for local variables is that Ok?
(please note that I have modified the code to have the bltouch support & added the filament empty sound as like it was in the original anycubic firmware.)

For the filament empty sound I have added this at the AnycubicTFT.cpp file:

at line 130:

`void  FilamentLack(){
    buzzer.tone(150, 3136); // G#7
    buzzer.tone(150, 2960); // FS#7
    buzzer.tone(150, 3520); // A#7
`

And added at line 749 this:

`  // since this is inside a loop, only set delay time once
      if (FilamentSetMillis) {
		  FilamentLack();
        #ifdef ANYCUBIC_TFT_DEBUG
          SERIAL_ECHOLNPGM("DEBUG: Set filament trigger time");
        #endif
        // set the delayed timestamp to 3000ms later
        fil_delay = fil_ms + 3000UL;
        // this doesn't need to run until the filament is recovered again
        FilamentSetMillis=false;
      }`

davidramiro, 2019-12-10:
Yes, 1445 bytes is fine.

shaddow501, 2019-12-10:
Ok

shaddow501, 2019-12-10:
I guess we can close this issue.

closed 2019-12-10 by shaddow501

belotv, 2019-12-11:
@shaddow501, for the filament runout sound, you may rather put it before ANYCUBIC_SERIAL_PROTOCOLPGM("J15"); //J15 FILAMENT LACK

In the loop where you put it, you do not allow the printer to recover by itself in 3s so you may have some false positive alarms.