davidramiro/Marlin-Ai3M-2.0.x #3: LV8729 Stepper driver

shaddow501, 2019-12-13:
Dear David

Maybe you can give some information.

I recently ordered the LV8729 motor  driver since I have heard good review about them.

I have installed them on the Z, Z1 & E0.
adjusted their voltage to about 1.100V.

Also made the modifications in the code:

#define Z_DRIVER_TYPE  LV8729 
#define Z2_DRIVER_TYPE LV8729 
#define E0_DRIVER_TYPE LV8729 

Also needed to switch the direction on the Z, and E0.
#define INVERT_Z_DIR true
#define INVERT_E0_DIR false

The direction now works fine the problem is that the moments is very slow and I also thing it is not the full movement.

My speculation that since it is by default 1/128 and not 1/16 could maybe cause this issue, but what I dont understand is why since if you define the driver type to be LV8729 it should know that that the stepping is 1/128 and work accordingly
[image]

do you have any clue?

davidramiro, 2019-12-14:
You will probably have to calibrate your steps. I have no experience with those kind of drivers so I can't really guide you through it but it seems probable that the higher microsteps need adjusted calibration of the axis steps per unit setting.

shaddow501, 2019-12-14:
Well I am not sure that it will solve the speed problem...
I think that the best thing to do is to use those drivers as 1/16 configuration and that deals with some hardware tweaking... 
Remove pins MS1 & MS2 that will not get high value from the board socket (and not burn the board) and solder them to the GND pin...

Well that is the idiot way lol, just wanted to see how 1/128 bit of a driver will bring me into better printing, but as today we get very nice printing also from 1/16 drivers configuration that I even not sure if this upgrade to 1/128 or some use 1/256 will be noticeable in the printed model.

davidramiro, 2019-12-21:
I only have really limited knowledge about microstepping, but what I do know for sure is that 128 or even 256 microsteps would be much more taxing to the puny little 8bit MCU on our Trigorilla. I'm not sure if it's up for the task.

To quote this thread on [StackExchange](https://3dprinting.stackexchange.com/questions/7572/0-9-motors-for-delta-printers):
>Also remember that if you are using 8-bit electronics (you hint to an ATMega board), then even 1/32 micro-stepping burdens the processor to achieve reasonable travel speeds. With 8-bit electronics, it is usually suggested to use 1/16 stepping.

shaddow501, 2019-12-21:
Than I guess I will take the LV8729 remove pins MS1 & MS2 that will not get high value from the board socket (and not burn the board) and solder them to the GND pin...
Then it will be 1/16. (cause now it is 1/128 buy default)

closed 2020-01-02 by davidramiro