davidramiro/Marlin-Ai3M-2.0.x #6: [BUG] Is using inject_P() to send commands to the command buffer from the anycubic_TFT.cpp file actually the right way to do this

hub-jba, 2019-12-27:
### Bug Description

Understanding that issue #4 will actually make the anycubic_TFT.cpp code command queuing work the same way in Marlin 2.0 as it did in Marlin 1.1.9 it's still worth considering if this is actually the correct way to do this.

Anytime we are injecting commands to the queue (aborting the current queue) we are technically throwing away commands that had been queued in the command queue just prior to our call.

This means that even in the Marlin 1.1.9 code version of the filament runout code exhibits this behaviour. And effectively once the print resumes we could in theory lose whatever commands were in the queue. Granted this is probably a small number of commands but still possible.

What we really need is a way to prevent the queue from accepting any more commands, but still enqueue our commands immediately after the last command in the queue.

Note sure if Marlin has a concept to do this. We need to review the key code to understand if this is even a problem, and if there is even an elegant way to solve this.

davidramiro, 2020-07-12:
Sorry, this should not have been closed. Some GitHub workflows were pulled from upstream that are a bit too quick for this repo.

I have merged your ExtUI repo code onto a new branch of this repo. CI reports that it wouldn't compile yet, will look into that later on.

hub-jba, 2020-07-12:
I have it compiling and working with absolute minimal changes to actual bugfix2.0 branch of Marlin.

Currently it's one change in the m125 code to wait for confirmation if using extensible UI. 

I am still tweaking the pause/timeout/resume behaviour as well as the filament runout behaviour, but everything else is working fine.

hub-jba, 2020-07-15:
On this particular issue much of the marlin code uses inject_P to inject commands around the pause functionality so i think it's okay to close this. There are also enqueue and enqueue_now methods, but since the rest of the code base uses inject_P we should be good.