davidramiro/Marlin-Ai3M-2.0.x #7: trigorilla stepper assignment Z2/E1?

Pandoriaantje, 2019-12-28:
I'm running a mixed (tmc2209/A4988) stepper driver configuration (A4988 on extruder). and i am very confused about configuring the steppers.

On a Trigorilla board, the second Z-axis is connected to E1 header

in "Configuration.h" I have 
`#define X_DRIVER_TYPE  TMC2209_STANDALONE
#define Y_DRIVER_TYPE  TMC2209_STANDALONE
#define Z_DRIVER_TYPE  TMC2209_STANDALONE
#define E0_DRIVER_TYPE A4988
#define E1_DRIVER_TYPE TMC2209_STANDALONE
`
But when i look into "pins_TRIGORILLA_14.h", E1 is not defined. and Y2 and Z2 have the same pins assigned to them (which actually match E1 header on the board). Mind you, there is no Y2 on a trigorilla, and the second Z is hardwired in parallel).

So should  "Configuration.h" have 
`define Z2_DRIVER_TYPE  TMC2209_STANDALONE`
or
`define E1_DRIVER_TYPE  TMC2209_STANDALONE`
if you have a 2209 driver on the second Z-axis driver?

in vanilla Marlin 2.0, this seems to have been changed, and "pins_TRIGORILLA_14.h" has an include for "pins_RAMPS.h", which has the correct pin assignments for the X, Y, Z, E0 and E1 headers on the trigo board.

davidramiro, 2019-12-28:
Confusing indeed, I wasn't so sure myself since the pin config comes from Anycubic themselves. You are correct, it should be Z2 according to the pins file. I simply commented E1 to be edited as well because on the silkscreen, it is (as I now know, incorrectly) actually the second Z-axis.

So yeah, regarding the configuration, Z2_DRIVER_TYPE should be what you want to define, although it doesn't matter if you define unused drivers so you could leave E1 defined as well.

Pandoriaantje, 2019-12-28:
It is something to keep in mind, because in vanilla marlin 2.0 this has been fixed, and now matches the actual pcb layout/silkscreen, so the Config.h will have to be edited according, when that gets merged.

akula3, 2020-02-28:
Hi Pandoriaantje,
I have TMC2209 V3.0 with corrected pinout labelling. Before I will do something wrong and/or damage something, I want to kindly ask you: how did you get working your TMC2209 drivers on Trigorilla board? Does it require some additional modifications like wiring, cutting pins, soldering etc.?
Is it configurable only for STANDALONE mode or UART is also possible and how, if yes?
Thanks a lot in advance!

Pandoriaantje, 2020-03-03:
> Hi Pandoriaantje,
> I have TMC2209 V3.0 with corrected pinout labelling. Before I will do something wrong and/or damage something, I want to kindly ask you: how did you get working your TMC2209 [...]

Being an 8-bit board, i wouldn't recommend UART, but it should be possible. You'dd need to at least cut the MS3 pin, if you want to run in stealthchop mode. MS3 is always high on trigorilla (no configuration jumpers), so it would run in SpreadCycle mode. You might also want to cut the 2 UART pins. To stepper motors can be reversed in firmware, so no need to manually reverse the stepper motor wiring.

akula3, 2020-03-03:
Thanks for your reply, Pandoriaantje!
I just installed my TMC2209v3 with cutting M3 on axes drivers and configured in standalone and reversed motor settings in the firmware. Movement works fine, real prints were not tested so far.
UART is an interesting option, but it would be difficult to realize due to a lack of free ports on Trigorilla.

AndiP1234, 2020-06-06:
I use the Fystec TMC2209 v3.0 without cutting any pin and without UART on my Mega S.
I can´t find the MS3 pin at the silkscreen.. Is this a problem? _Marlin 1.1.9 (TMC2208 Standalone Version)_

akula3, 2020-06-06:
AndiP1234,
MS3 Pin is the SP (Spread) Pin.
If you do not cut it, the driver will run in Spread Cycle mode; if you cut it, the driver will run in StealthChop mode (noisy). Info from Fysetc: "SP Chopper mode selection: Low/pin open=StealthChop, High=SpreadCycle".
I'm not sure, if TMC2209 will work correctly with Marlin 1.1.9, because it Configuration.h has no option for TMC2209.
[image]

quattroerik, 2020-06-07:
I have the new TMC2209 too, no changes on the wiring of the connector. When having the MS3/SP PIN connected i had a loud hissing noise (in german: rauschen) when the motors were idle. Now i have removed it for the x/y/z axis, but kept it for the E0. Works fine with the precompiled TMC2208 version from David. Supersilent, except of the noctua fan which runs at 100% :-)

AndiP1234, 2020-06-07:
is the noise difference big with cutting the pin?

quattroerik, 2020-06-07:
Yes, with the PIN I felt something is wrong. I unsoldered the PIN, you can
then even put it back. Without I can not hear any noise.. (just the
extruder made a loud clicking noise when pulling back the filament.)

AndiP1234, 2020-06-07:
Danke für die Hilfe. Selbst im Spread Cycle sind die Treiber schon wesentlich angenehmer als die Standard.
Ich wollte eben dein Video kommentieren. Geht aber iwie nicht. Vielleicht kannst du noch eine Videobeschreibung hinzufügen. Ich hatte vor den Kauf ewig nach einer Anleitung gesucht, bzw ob die Treiber überhaupt mit dem Board laufen.

Short instruction for other makers (based on https://github.com/davidramiro/Marlin-Ai3M-2.0.x/issues/7)
Fystec TCM2209 v3.0 for all 5 Motors. Pin SP/MS3 is cut off for the axis drives. Only E0 with the PIN -> Spread Cycle. (Removed PIN = StealthChop)
https://user-images.githubusercontent.com/58461189/83955874-11f3cc00-a858-11ea-9d91-8e04bfb97954.png
Runs smooth with Davids precompiled TMC2208 Version. (https://github.com/davidramiro/Marlin-Ai3M-2.0.x)

I adjust Vref to 0.95V for the axis and 1.1V for the extruder.

quattroerik, 2020-06-07:
Gerne. Habe ein paar mehr Infos zum Video hinzugefügt und auch hier her verlinkt. Kommentare sollten jetzt eigentlich auch gehen.
Btw. mein VRef: X/Y/Z/E1: 0.93V, E0: 0.8V
Den Plan die mit UART anzusteuern habe ich erstmal verworfen da ich dafür die FW anpassen muss sowie die Verbindungen. (Anschluss Diagram gibts hier: https://ibb.co/Q86y3pb)

Welcome. Added information to the video.
Droppen the plan to connect with UART due to FW adjustments as well as required connections.

akula3, 2020-06-14:
> Den Plan die mit UART anzusteuern habe ich erstmal verworfen da ich dafür die FW anpassen muss sowie die Verbindungen. (Anschluss Diagram gibts hier: https://ibb.co/Q86y3pb)

Hallo **quattroerik**,
Wenn ich richtig verstehe, hier werden zwei serielle Ports mit 2 und 3 Adressen benutzt.
Die Theorie ist soweit klar, passt auch zu Info in Configuration_adv.h:
   ```
1* Four TMC2209 drivers can use the same HW/SW serial port with hardware configured addresses.
   2* Set the address using jumpers on pins MS1 and MS2.
   3* Set *_SERIAL_TX_PIN and *_SERIAL_RX_PIN to match for all drivers on the same serial port, either here or in your board's pins file.
```
In der Praxis ist mir unklar, welche Ports  sollte man für 1* nehmen (ob überhaupt).
Vermutlich, wird es in der pins_RAMPS.h bei den Parametern #define [Treiber]_HARDWARE_SERIAL Serial[Port Nummer] definiert.
Nur welche PINs liegen an welchen Ports bei Trigorilla?
Das mit 2* ist klar, wird in Configuration_adv.h definiert.
Bei 3* würde ich sagen, für *_SERIAL_TX_PIN können irgendwelche freie PINs genommen werden, für die *_SERIAL_RX_PIN dürfen nur interrupt capable PINs genommen werden (z.B. D12). Das wird in der pins_RAMPS.h definiert. Wobei *_SERIAL_RX_PIN is optional und kann ausgelassen werden (-1), wenn man keine Rückmeldung vom Treiber braucht.
Hast du für das Bild auch ein Tutorial für Einstellungen in der Firmware?

Pandoriaantje, 2020-06-14:
> AndiP1234,
> MS3 Pin is the SP (Spread) Pin.
> If you do not cut it, the driver will run in Spread Cycle mode; if you cut it, the driver will run in StealthChop mode (noisy). Info from Fysetc: "SP [...]

it should work fine with 2208 config (in standalone mode). if you select tmc2208_standalone, it defines "HAS_TRINAMIC_STANDALONE 1", which in turn defines some timing/rate settings, for 2208/2209 drivers. "MINIMUM_STEPPER_POST_DIR_DELAY 60", "MINIMUM_STEPPER_PULSE 0", MAXIMUM_STEPPER_RATE 5000000.