davidramiro/Marlin-Ai3M-2.0.x #12: [FR] Mega X Support

DunklerPhoenix, 2020-03-09:
Heho.

Will you support the Anycubic Mega-X?
What do you need for informations?

Here I posted the M503: https://www.reddit.com/r/anycubic/comments/fcqz6f/anycubic_megax_configurations_m503/

and here you get the firmwares:
https://www.reddit.com/r/anycubic/comments/fcsy92/anycubic_megax_stock_firmware/

The printer is nearly the same as Mega-S except of the size of X Y and Z and the acceleration rates.

Greetings

davidramiro, 2020-03-09:
Probably not a big deal porting it, but it's hard for me without owning the device. From a first glance, the screen firmware changed. Hopefully they didn't change they way it communicates.

I am sure somebody who owns a Mega X can fork this firmware and get it to run.

DunklerPhoenix, 2020-03-09:
Yeah the screen is a problem. I flashed a Mega-S Firmware for testing and the display didn't change a bit. Unfortunately I dont have enough programming experience to create a new interface ^^

davidramiro, 2020-03-09:
The display interface doesn't change, it's hardcoded. It's just the communication between the screen and the board that this custom Marlin version handles. Maybe you can try and see if some of the functions of the new screen still work with this firmware?

leberschnitzel, 2020-05-07:
I've flashed with this one.
Customized "configuration.h" it to bed size on line 888 + 889 + 897
Commented 559 to 566 to "stock drivers" (if someone knows what drivers are used, I'm happy to change.
Needed to reverse, so also changed lines 857-865 according to reverse comments.

Testing on screen: 
Heating bed works.
Heating Nozzle works.
Motor off works.
Home X works.
Home Y does seem to move the bed back instead.
Home Z works.
Move Axis: Z Works, Y /X don't.
Special Menu works.

My guess is I just need the right drivers for X / Y and then I can try again.

leberschnitzel, 2020-05-12:
I tried a bit more but sadly no success so far.
Does someone have an idea in what direction I could test to get the Y axis working correctly?
I think now it might have to do with the new endstop, which is light based.

Kit613, 2020-05-12:
leberschnitzel  mail me  kit613@o2.pl

Kit613, 2020-05-18:
https://www.thingiverse.com/thing:4358011?fbclid=IwAR1VRPXaUGK98Lnl9ldNtQuv77tKaBSBPdUHT9QsI9JdIjhIh1ahsjY_6As

jonatanpolak, 2020-05-27:
> https://www.thingiverse.com/thing:4358011?fbclid=IwAR1VRPXaUGK98Lnl9ldNtQuv77tKaBSBPdUHT9QsI9JdIjhIh1ahsjY_6As

Yes, this works. However, this is Marlin 1.1.9 version. And there was so many changes in version 2.0 that I would like to see them in my printer :)

I have one question:
Printer Anycubic i3 Mega X 2020 - official firmware, https://github.com/ANYCUBIC-3D/MEGA_X_V1.2.6/blob/master/Configuration.h#L139
MOTHERBOARD BOARD_RAMPS_14_EFB

There has been a change and it is no longer BOARD_TRIGORILLA_14 in mega X, right?