davidramiro/Marlin-Ai3M-2.0.x #18: [BUG] MAXTEMP trigger when SD-Card print ends

SimGas, 2020-04-08:
### Bug Description
I discovered a bug while printing using the SD-Card feature in the software.

My printer ran fine on your previous Marlin version (1.1.9)

After the upgrade to Marlin 2.0, my printer does not finish print jobs from SD card correctly. 
Hoever, When I print using USB Connection, every print works just as expected.

When I monitor the Serial output I get the following response:

`SENT: M105
READ: X:86.81 Y:94.96 Z:1.53 E:0.00 Count X:8138 Y:8138 Z:612
READ: ok T:219.79 /220.00 B:74.99 /75.00 @:47 B@:62
READ: X:92.47 Y:88.40 Z:1.53 E:0.00 Count X:7010 Y:7498 Z:612
READ: X:94.28 Y:86.81 Z:1.53 E:0.00 Count X:7398 Y:7072 Z:612
SENT: M105
**READ: TFT Serial Debug: SD print done... J14
READ:  DEBUG: AI3M Pause State: 0
READ: Error:MAXTEMP triggered, system stopped! Heater_ID: 0**
SENT: M105
READ: Error:Printer halted. kill() called!
READ: TFT Serial Debug: Kill command... J11`
 
After that, the printer immediatelly stops.

I am quite sure the connection to the hotend thermistor is not loose. 

Any idea what could trigger the kill function?

### My Configurations

Link to configuration file: [https://cloud.hansen-its.de:443/share.cgi?ssid=0O2KKcb](url)
### Steps to Reproduce
Error does not occur while printing via usb (Simplify 3D)
However, the same G-Code does not work when I put it on to SD-Card 
Start print using SD-Card 

Last few Lines of the G-Code are:

`G92 E0.0000
G1 E-7.0000 F3900
; layer end
G91; relative positioning
G1 Z5 ;lift Z axis after printing
G90; absolute positioning
G1 X0 Y195 F2400 ; position for easy part removal
M106 S0 ; fan off
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
M84 ; disable motors`

The print finishes as expected, but the printhead stays at the object (it does not lift the nozzle after printing) and the build plate is not moved forward. 

**Expected behavior:** [What you expect to happen]
Printer finishes print and does all commands as expected
**Actual behavior:** [What actually happens]
Printer calls the kill function and stops immediatelly
#### Additional Information

SimGas, 2020-04-08:
I am sorry, but I just discovered one cause of the MAXTEMP trigger. 
I did not reset the EEPROM after upgrading the firmware. 

I have done that now. That caused the MAXTEMP trigger to not occur anymore. 

However, Now I get a complete restart of the printer after each print job:

`
`[...]
SENT: M105
READ: X:86.81 Y:94.96 Z:1.53 E:0.00 Count X:8138 Y:8138 Z:612
READ: ok T:220.13 /220.00 B:75.00 /75.00 @:42 B@:61
READ: X:92.47 Y:88.40 Z:1.53 E:0.00 Count X:7010 Y:7498 Z:612
READ: X:94.28 Y:86.81 Z:1.53 E:0.00 Count X:7398 Y:7072 Z:612
SENT: M105
READ: TFT Serial Debug: SD print done... J14
READ:  DEBUG: AI3M Pause State: 0
READ: echo:SD card ok
READ: TFT Serial Debug: SD card inserted... J00
READ: DEBUG: Set filament trigger time
READ: start
READ: echo:Marlin 2.0.4.4
Marlin 2.0.4.4
READ: 
READ: Marlin-AI3M v2.0.1
[...]`

davidramiro, 2020-04-11:
Do you have your filament sensor connected?

SimGas, 2020-04-11:
Yes, the filament sensor is still connected. Electrical-hardware-wise my
printer is still unmodified.

davidramiro, 2020-04-11:
Interesting. Mine is not connected and the printer does not restart after an SD print. The second debug message implies it gets triggered.

I have to test that sometime, but please excuse my response time. I don't have much time to spend on this project at the moment.

SimGas, 2020-04-11:
If you want me to test something, just tell me.
The bug is currently not a problem to me as I started using octoprint
anyway :-)

closed 2020-07-12 by github-actions[bot]