davidramiro/Marlin-Ai3M-2.0.x #19: [BUG] (short description)

Nissal1983, 2020-04-11:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

davidramiro, 2020-04-11:
Did you change some BLtouch settings or similar before trying to compile?

Nissal1983, 2020-04-11:
Ich glaube ich kann deutsch schreiben oder?? 
Nein, ich habe soweit noch keine Settings geändert. Ich habe manuel levling deaktievert, Mesh levling deaktiviert und Auto bed levling biliniar aktiviert. Als Probe nozzle as Probe aktiviert. Dann kommt der Fehler. Wenn ich einen bl Touch aktiviere kann ich kompilieren, auch geht es wenn ich das Orginal Display deaktivierte und nur ein reprap Display aktiviere. Er bringt den Fehler nur bei nozzle as Probe oder fix mounted Probe wenn das Anycubic Display aktiviert ist. Sonst hab ich keine Änderung gemacht.

davidramiro, 2020-04-11:
Hab den Fehler gefunden. Gefixt in commit d47dbd5e4818d82c0353f178c58276aac77873b8.

closed 2020-04-11 by davidramiro

akula3, 2020-04-14:
Hallo David,

bei mir ist #define BLTOUCH aktiviert, da ich einen BLTouch Sensor für AUTO_BED_LEVELING_BILINEAR benutze.
Wird sich die Änderung in commit d47dbd5 für meinen Fall negativ auswirken?

davidramiro, 2020-04-14:
Nein, kein Problem. Der Code dort war eigentlich schon lange überflüssig.