davidramiro/Marlin-Ai3M-2.0.x #20: [BUG] Out of filament sensor does not trigger printing pause

gmag11, 2020-04-14:
### Bug Description

Filament sensor does not pause print. When filament is out of sensor a message is shown on screen but printing goes on normally.

### My Configurations

[Marlin.zip](https://github.com/davidramiro/Marlin-Ai3M-2.0.x/files/4476757/Marlin.zip)

### Steps to Reproduce

1. Put a piece of filament on sensor to simulate run out.
2. Start printing as usual
3. Remove filament from sensor

**Expected behavior:** I would expect that printing pauses waiting for new filament

**Actual behavior:** A message is shown on LCD, so sensor works fine. Printing process goes on.

davidramiro, 2020-04-14:
This should be caused by the issue described here: https://github.com/davidramiro/Marlin-Ai3M-2.0.x/issues/5

Sorry, I haven't had the time recently to take a deeper look at this. I am quite busy with work these days, so I'll still have to postpone thoroughly testing this for a couple of weeks.

davidramiro, 2020-04-14:
Just to confirm, you are printing from SD card, right?

gmag11, 2020-04-14:
No, I am printing from Octoprint. Is that the reason maybe?

davidramiro, 2020-04-14:
Yup, that is indeed the reason, the printer has no way of letting OctoPrint know that filament has run out, natively. You could install a filament sensor plugin and connect the filament sensor directly to the Pi, that would work well.

gmag11, 2020-04-15:
I get this in OctoPrint's terminal:
```
Recv: TFT Serial Debug: Filament runout... J15
[...]
Recv: TFT Serial Debug: Filament runout recovered
[...]
Recv: DEBUG: Set filament trigger time
[...]
Recv: DEBUG: 3000ms delay done
Recv: TFT Serial Debug: Filament runout... J15
```
Could this be used to tell it to pause printer and park noozle?

davidramiro, 2020-04-15:
You could write a plugin that detects this and stops the print, just not natively.

closed 2020-05-08 by davidramiro