davidramiro/Marlin-Ai3M-2.0.x #21: [FR] Include Piezo Build

psi-4ward, 2020-04-17:
Hi David,
first: thanks for your work!

I've some questions:
* Would you like to include a builds which support the piezo sensor for automatic leveling? (If yes I could prepare PR)
* You've a travis integration but seems travis does not really build the FW, why are you going this way?

Thanks and greets

davidramiro, 2020-04-17:
Hi there!

I could set up a branch for piezo support where you can PR onto, but I won't do any builds/releases of that as I can't test it (same with BLtouch).

Yeah, indeed, travis is broken atm, it worked a [few commits ago](https://travis-ci.org/github/davidramiro/Marlin-Ai3M-2.0.x/builds/656650673) but I think something broke when merging from upstream. I don't have much time to have a look at this at the moment, sorry...

psi-4ward, 2020-04-17:
Ah I see. So the better way would be a fork of your repo to release the piezo builds by someone who can test the stuff.

closed 2020-04-17 by psi-4ward

davidramiro, 2020-04-17:
Yeah, thanks for your understanding. Just as there is a BLtouch fork of Marlin-Ai3M, this would be basically the same then.

This might be of use for you: https://github.com/kulfuerst/Marlin_Ai3M_2.0.x_bugfix_piezo

I helped some guy set up my 2.0 repo with a piezo. It's a bit dated in terms of commits but maybe you can get something out of it!