davidramiro/Marlin-Ai3M-2.0.x #22: [BUG] Hot end fan runs only at 100%

gmag11, 2020-04-28:
### Bug Description
Does hot end fan support PWM control? It only blows when fan speed is set to 100%. In any other value fan keeps stopped.

### My Configurations

[Marlin.zip](https://github.com/davidramiro/Marlin-Ai3M-2.0.x/files/4548705/Marlin.zip)

### Steps to Reproduce

1. Use slicer to configure fan to be active at different speeds
2. Send gcode to print

**Expected behavior:** Fan should adapt its speed to setting

**Actual behavior:** Fan works only if speed is set to 100%

#### Additional Information

Fan I use is a 5015 like this: https://es.aliexpress.com/item/4000011565156.html

davidramiro, 2020-04-28:
It's right there in the FAQ... 

https://github.com/davidramiro/Marlin-AI3M/wiki/Frequently-Asked-Questions#my-sunon-or-other-custom-fan-only-runs-at-100

https://github.com/davidramiro/Marlin-AI3M/wiki/Customization-&-Compiling#PWM-Settings

closed 2020-04-28 by davidramiro

gmag11, 2020-04-29:
Thank you. PWM signal is OK. Using stock blower works fine. I will read docs regarding PWM tunning