davidramiro/Marlin-Ai3M-2.0.x #23: [BUG] (y_min & x_max endstop not detected)

Buri2894, 2020-05-05:
### Bug Description

When triggered, both the y_min & x_max endstop do not change state

### My Configurations

I'm running latest 2.0.x commit [9b7f0ba](https://github.com/davidramiro/Marlin-Ai3M-2.0.x/commit/9b7f0bae9180a6f73e94154505e08b7ce7e275a8), with stock code, no modifications
Moved from Marlin 1.1.9 and there was no problem on that firmware.
Tried to compare both configuration file in the "Endstop" section" but without any result

**Expected behavior:** [Normal Homing of all Axes]

**Actual behavior:** [Y Axis and the second Z axis keep pushing the rail after reaching the endstop]

davidramiro, 2020-05-05:
Thank you in for reporting this. I'll have a look in the evening.

When upgrading, please keep in mind that this repository is in early beta status.

Buri2894, 2020-05-05:
Than you for your support.
I know that's in early beta but when I first downloaded the whole repository 4-5 days ago, compiled with just the TMC2208 flipped direction, everything was working fine. Then I had to restore back o Marlin 1.1.9 to check some values and (my fault) i deleted the previous 2.0.x folder so when i upgraded again yesterday I found this problem on my printer

Dalton0190, 2020-05-05:
In "pins_TIGORILLA_14.h" they are not defined and the default pins in the included "pins_RAMPS.h" are different

Nissal1983, 2020-05-06:
The problem is an incorrect pin in pins_RAMPS.h
To fix it, in the file pins_RAMPS.h change the pin for y min from pin 14 to pin 42 then it works

davidramiro, 2020-05-06:
Sorry, I overwrote the pin config when pulling from the Marlin repo. Fix incoming in a few minutes!

Shnoo, 2020-05-06:
Are you sure this was the right fix? now none of my end stops work anymore

davidramiro, 2020-05-06:
Ugh. Thanks for reporting. I can't test it atm, I am not at my printer right now.

Could you try this one? 
[pins_TRIGORILLA_14.zip](https://github.com/davidramiro/Marlin-Ai3M-2.0.x/files/4588156/pins_TRIGORILLA_14.zip)
Goes into /src/pins/ramps/

Shnoo, 2020-05-06:
Forget what i just wrote. I recompiled the version again to double check and it seems to work now. No idea what went wrong. 
But thank you for your great work and fast responses.

davidramiro, 2020-05-06:
No worries. Just a weird glitch maybe. I'm still leaving this open for some more time, if someone else reports it, we know it wasn't you.

Dalton0190, 2020-05-07:
Latest version works for me

commit 157f38f6 (2020-05-06): Fix endstop pin config

Resolve #23
closed 2020-05-08 by davidramiro