davidramiro/Marlin-Ai3M-2.0.x #26: Bug: "//#define ARC_SUPPORT" in "conv._adv.h" create compiler error

NetrunnerAT, 2020-05-25:
Servas David!

wenn man ARC_SUPPORT deaktiviert bekommt man einige compiler errors. Hast du was in der planer routine geändert?

```
C:\Users\TheDude\AppData\Local\Temp\cc4G2ake.ltrans1.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':

<artificial>:(.text+0x6d14): undefined reference to `plan_arc(XYZEval<float> const&, XYval<float> const&, unsigned char)'

collect2.exe: error: ld returned 1 exit status
```

davidramiro, 2020-05-25:
Servus!

Ja, das wird hierfür verwendet: https://github.com/davidramiro/Marlin-Ai3M-2.0.x/blob/2.0.x/Marlin/src/gcode/temp/M888.cpp

Kleine GCode routine die ich gebastelt hab.

NetrunnerAT, 2020-05-27:
Ich schau mir das später mal an ... ist bisserl blöd, wenn man die Routine abschalten will. Die Compiler Meldung ist irgend wie schwer zu lesen. Du verwendest das um Kreisbewegungen zu machen? Um mit dem Bauteillüfter die Ultrabase zu kühlen?

davidramiro, 2020-05-31:
Korrekt. Du kannst M888 einfach aus dem GCode parser entfernen und dann sollte die Abhängigkeit nicht mehr bestehen.

closed 2020-05-31 by davidramiro