davidramiro/Marlin-Ai3M-2.0.x #29: [Question] should I change steppers pin before compile?

proof666, 2020-10-22:
Hi, I’m a little confused about pins settings in trigorilla_14.h and ramps.h

I see that 1.x version has configuration for z2 stepper in
https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/pins_TRIGORILLA_14.h which corresponding for actual wirings. But in 2.x version I see that e1 has pins that actually wired to z2.

Should I change this setting before compile 2.x version?

And another question is if I want to add additional extruder, what should I change in this files? Is enough to just add e2 block with corresponding pins?

davidramiro, 2020-10-23:
E1 and Z2 are kinda used interchangeably since that is what's marked on the Trigorilla board. By default, the second Z motor plugs into the E1 pins on the board, hence it's configured that way. No, you don't need to change anything before compiling, it's just a small naming difference.

Yes, adding E2 and enabling its use in `Configuration.h` and `Configuration_adv.h` should work.

proof666, 2020-10-28:
Can you point me to place in config where second z stepper mapped to E1?

davidramiro, 2020-10-28:
So, just so I understand, you are trying to power a second extruder via the motor from the second Z-axis, right?

proof666, 2020-11-02:
I want save separate configuration for both z steppers and use pins under ext1/ext2 for additional extruder.