davidramiro/Marlin-Ai3M-2.0.x #30: [QUESTION] I'm new, do you suggest this or v1.

RedS-DEV, 2020-12-01:
This is in beta, so do you suggest using that or v1.

davidramiro, 2020-12-01:
Use the 1.1.9 version if you're not comfortable troubleshooting potential issues.

closed 2020-12-01 by davidramiro

RedS-DEV, 2020-12-01:
Oh ok, can i enable opening sound on v1 and will power loss resume came?

davidramiro, 2020-12-01:
Yes, you can enable it in the source code. Uncomment https://github.com/davidramiro/Marlin-Ai3M/blob/master/Marlin/Configuration.h#L1524 and compile.

No, power loss resume is not available on any of those two firmwares.

RedS-DEV, 2020-12-01:
Ok, after sending firmware from cura, what should i do? Close and reopen printer?

davidramiro, 2020-12-01:
https://github.com/davidramiro/Marlin-Ai3M/wiki/Customization-&-Compiling

Compile via these instructions, flash the hex file, send M502 and M500. Done.