davidramiro/Marlin-Ai3M-2.0.x #33: [ARC SUPPORT not true] (Failure with G2/G3)

FM-500, 2021-02-09:
### Bug Description

Installed to Hex File on a Anycubic Mega-S in wanted to use the Arc Welder plugin which needs Arc Support.
G2/G3 commands give out an error, which should mean that Arc Support isnt enabled at the firmware.
I checked the Code in the Configuration_adv.h file at line 1631 and it looks like it should be enabled.
Are there any additional steps, that need to be made?

FM-500, 2021-02-09:
[Same issue for reference, seems to be solved?](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/38#issuecomment-733188664)

FormerLurker, 2021-02-11:
What version of Marlin is this fork based on?  There have been quite a few updates to the interpolation algorithm recently that fix several glitches.

FM-500, 2021-02-11:
> What version of Marlin is this fork based on? There have been quite a few updates to the interpolation algorithm recently that fix several glitches.

I am only able to see that the base is something like 2.0.x, which isn't really helpful.
I found out though that a blank G2/G3 command leads to a connection error.
However if I specify a movement with G2/G3, it is carried out correctly.

The compressed Gcode is printed pretty good, but compared to the original, there are slight imperfections.
Gcode looks weired in gcode.ws as well.

FormerLurker, 2021-02-11:
I don't want to get too much off-topic, but most viewers don't support arcs.  Those that do often have issues, like low resolution.  Try simplify 3d or the PrettyGcode viewer in OctoPrint.

The imperfections could definitely be due to missing fixes in the upstream branch. If I have some time I'll take a look at the version history for the interpolation code and will see what I can find.

FM-500, 2021-02-11:
Right, S3D showed the Gcode correctly, according to the corresponding STL.
I must admit though, that I probably did not choose an ideal model to test arcs.
It's a mostly suqare nameplate with letters in it.
What model would be suitable to make some tests?

FormerLurker, 2021-02-11:
> What model would be suitable to make some tests?

The letters should be an OK test.  I like using (surprise surprise) benchy to test quality, but some of the detailed miniatures of people/superheroes/monsters, etc are typically a good torture test since they are mostly random organic shapes.

FM-500, 2021-02-11:
I will print two Benchys. One without Arcs, another after Arc Welder applied.
Am I supposed to actually see a difference when viewing the Gcode in S3D?
Because I think I cannot.
Benchy gets compressed by about 38%, but the toolpaths look pretty much the same to me.

[image]
NO ARCS
[image]
AFTER PROCESSING WITH AW

FormerLurker, 2021-02-11:
> Benchy gets compressed by about 38%, but the toolpaths look pretty much the same to me.

This doesn't appear to be related to this issue.  However, the whole point of arc welder is that you shouldn't be able to tell the difference between the toolpaths of the welded and original file.  This may or may not translate into quality improvements.  However, if the firmware is bad at generating arcs, then the quality will suffer.