davidramiro/Marlin-Ai3M-2.0.x #35: Update to latest Marlin 2.0.7?

MrBill1234, 2021-02-28:
Is this project going to be updated to the latest marlin?

Thanks.

ashleysommer, 2021-03-08:
I was wondering the same thing. For a long time this project was the only custom Anycubic Mega firmware available, and I'm still on an earlier version.

Luckily, I found [this repo by Knutwurst](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S) that appears to have equaled, and even surpassed the features of davidramiro firmware. And is based on Marlin 2.0.6.1 (and I think master branch is on 2.0.7).
Also found [files from Bavarian-Chris](https://www.thingiverse.com/thing:4358979) on Thingiverse, which might be the same files built by knutwurst.

bassamanator, 2021-04-03:
This work takes a long time and I suspect it's a thankless job. Hopefully it will get updated but for now, I'm grateful for the work that David has done. Thank you!

I hope to help in the testing process if the need arises.

davidramiro, 2021-05-30:
Sorry for my unresponsiveness. I am currently working on rebasing the whole project onto the current 2.0.8 of upstream Marlin. This will take a bit of time, as it will be using the new ExtUI feature. I will have to reimplement some features like M888 and have to catch up on quite a lot of updates. While I have been inactive, the Marlin team has chugged out some quite significant improvements, that I too do not want to miss out on. Hope I can finish this soon.

MrBill1234, 2021-05-30:
Sounds great.  Do you have an idea on approximate timeframe?  Also please be sure to include the MeatPac functionality if it is not in the tree you have forked from 👍

sjscicluna83, 2021-11-19:
When you update could you please enable Host Action Commands in Configuration_adv.h for your precompiled code this will allow Marlin and octoprint to work better together for things like Filament run out if it is still connected to the mainboard.
For more info see below.

https://youtu.be/fpphvl5C2Cw?t=651

https://reprap.org/wiki/G-code#Action_commands

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
#endif

skullmonkey911, 2023-01-16:
@davidramiro : Do you still plan to release this version based on Merlin 2.x or is the project dead?